pytroll/pygac #16: Unit test failure

avalentino, 2018-12-24:
Unit tests fail on linux (Ubuntu 18.10).

```
$ env PYGAC_CONFIG_FILE=etc/pygac.cfg python setup.py test

[CUT]

======================================================================
ERROR: test_calibration_ir (pygac.tests.test_calibrate_klm.TestKLMCalibration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/antonio/projects/pygac/pygac/tests/test_calibrate_klm.py", line 203, in test_calibration_ir
    line_numbers=np.array([1, 2, 3]))
  File "/home/antonio/projects/pygac/pygac/calibrate_klm.py", line 572, in calibrate_thermal
    tprt = np.zeros((float(number_of_data_records)))
TypeError: 'float' object cannot be interpreted as an index

======================================================================
FAIL: test_calibration_vis (pygac.tests.test_calibrate_klm.TestGenericCalibration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/antonio/projects/pygac/pygac/tests/test_calibrate_klm.py", line 76, in test_calibration_vis
    self.assertTrue(np.allclose(ref1, expected[0]))
AssertionError: False is not true

======================================================================
FAIL: test_calibration_vis (pygac.tests.test_calibrate_pod.TestGenericCalibration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/antonio/projects/pygac/pygac/tests/test_calibrate_pod.py", line 76, in test_calibration_vis
    self.assertTrue(np.allclose(ref1, expected[0]))
AssertionError: False is not true

----------------------------------------------------------------------
Ran 22 tests in 6.238s

FAILED (failures=2, errors=1)
Test failed: <unittest.runner.TextTestResult run=22 errors=1 failures=2>
error: Test failed: <unittest.runner.TextTestResult run=22 errors=1 failures=2>
```

sfinkens, 2019-03-25:
Fixed by #24

mraspaud, 2019-03-25:
Fixed

closed 2019-03-25 by mraspaud