pytroll/pygac #22: Thermal calibration error

sfinkens, 2019-02-22:
While processing 2017/18 data I noticed ~250 errors of the following type:

```
python bin/pygac-run testbench/NSS.GHRR.M1.D18337.S2304.E0005.B3222526.GC 0 0
[ INFO pygac.gac_reader 2019-02-22 15:36:34.498] Reading NSS.GHRR.M1.D18337.S2304.E0005.B3222526.GC
[ DEBUG pygac.gac_reader 2019-02-22 15:36:34.606] Removed 0 scanline(s) with corrupt scanline numbers
/cmsaf/nfshome/sfinkens/software/devel/pygac/pygac/geotiepoints.py:575: RuntimeWarning: invalid value encountered in arccos
  90 - rad2deg(arccos(z__/EARTH_RADIUS)),
/cmsaf/nfshome/sfinkens/software/devel/pygac/pygac/geotiepoints.py:578: RuntimeWarning: invalid value encountered in arcsin
  / EARTH_RADIUS))))
[ DEBUG pygac.gac_reader 2019-02-22 15:36:36.458] Corrected 0 timestamp(s)
Traceback (most recent call last):
  File "bin/pygac-run", line 104, in <module>
    reader(args.filename, args.start_line, args.end_line)
  File "/cmsaf/nfshome/sfinkens/software/devel/pygac/pygac/gac_klm.py", line 579, in main
    channels = reader.get_calibrated_channels()
  File "/cmsaf/nfshome/sfinkens/software/devel/pygac/pygac/gac_reader.py", line 254, in get_calibrated_channels
    self.spacecraft_name)
  File "/cmsaf/nfshome/sfinkens/software/devel/pygac/pygac/gac_calibration.py", line 477, in calibrate_thermal
    prt[ifix] = np.interp(ifix[0], inofix[0], prt[inofix])
  File "/cmsaf/nfshome/routcm/Modules_CentOS/python/2.7.14/lib/python2.7/site-packages/numpy/lib/function_base.py", line 2061, in interp
    return interp_func(x, xp, fp, left, right)
ValueError: array of sample points is empty
```

This seems to be yet another data corruption, because there are two very similar files with slightly different time stamps from different ground stations:

- [NSS.GHRR.M1.D18337.S2304.E0005.B3222526.GC](https://public.cmsaf.dwd.de/data/sfinkens/pygac/NSS.GHRR.M1.D18337.S2304.E0005.B3222526.GC)
- [NSS.GHRR.M1.D18337.S2308.E0005.B3222526.SV](https://public.cmsaf.dwd.de/data/sfinkens/pygac/NSS.GHRR.M1.D18337.S2308.E0005.B3222526.SV)

The errror occurs with the first file, the second one is processed just fine. Currently we would select the first one because it contains more data (according to the timestamps).

abhaydd, 2019-02-25:
Stephan,

Thanks for reporting this. I will start looking at it from tomorrow. Please feel free to fix anything you deem necessary. 

Cheers,
Abhay.

sfinkens, 2019-10-14:
@mraspaud I haven't looked further into it since then

PR #117 (2022-10-27): Fix calibration when 3b is totally deactivated
Fully deactivated 3b channel can crash the calibration. This PR fixes this.

This fixes #78 and fixes #22 in principle.
closed 2022-10-27 by mraspaud