pytroll/pygac #39: Update release in preparation for conda-forge package

djhoese, 2019-11-19:
The following things need to be or should be fixed in a patch release before I make a conda-forge package:

- [ ] The contents of the README.md file should be loaded in to the `long_description` parameter in `setup.py`. Since it is markdown you will also need `long_description_content_type='text/markdown'`.
- [ ] The README badges should be updated to point to the pytroll repository instead of @adybbroe's old version of this.
- [ ] The LICENSE file should be renamed `LICENSE.txt` (this is the default in conda-forge, not required to rename but eh why not). It should then be included in the `MANIFEST.in`.

After all of the above are complete:

- [ ] Make a new patch release

mraspaud, 2019-11-25:
Sounds like a good plan PR is welcome

djhoese, 2019-11-25:
This issue was my way of delegating the tasks to someone else :wink: 

I'm not really involved with this project but was going to make the conda-forge package to help where I could. I'm not sure I'll have time to do this soon unless I get bored and avoid my real work.

sfinkens, 2019-11-25:
I'll put it on my PCW list. Let's see how far I get :)

PR #101 (2022-01-10): Prepare for next release
Prepare for next release and conda-forge. Closes #39, closes #46.
closed 2022-01-10 by sfinkens