pytroll/pygac #43: Links to POD/KLM user guides are broken

sfinkens, 2019-11-28:
I've asked NOAA NCDC where the pages are located now.

mraspaud, 2019-11-28:
Thanks. I think @abhaydd had a place, but it's not any official website.

sfinkens, 2019-11-28:
http://webapp1.dlib.indiana.edu/virtual_disk_library/index.cgi/4284724/FID1385/podug/index.htm
http://webapp1.dlib.indiana.edu/virtual_disk_library/index.cgi/2790181/FID1497/klm/index.htm

gerritholl, 2019-11-28:
I recommend to get offline PDF copies.  I can't attach one here due to the 10 MB file limit but we don't know if the University of Indiana will keep this one forever.

gerritholl, 2019-11-28:
See #44

sfinkens, 2019-12-02:
Got a reply from NOAA: https://www1.ncdc.noaa.gov/pub/data/satellite/publications/podguides/

gerritholl, 2019-12-02:
That's where I got the PDFs from.  I decided to download a local copy.  I found that the US government has a habit of shutting down whenever I need to consult those guides.

PR #99 (2022-01-04): Improve documentation
Improve documentation:

- Update installation instructions
- Add usage examples
- Move algorithm descriptions to a new "Methods" chapter
- Move HDF description to a new "Legacy Output" chapter
- Add more readers and an inheritance diagram to API chapter

You can preview the updated version here: https://sfinkens.github.io/

Along the way I also included dependencies for development in `setup.py` and fixed a minor bug in `Reader.get_calibrated_channels` (using timestamps before they are computed).

Closes #85, closes #43.
closed 2022-01-04 by sfinkens