pytroll/pygac #49: Calibration Coeffs Patmos-X 2017

helgaweb, 2020-01-22:
I found some twisted numbers for the metopb coeffs in the calibration.py script.
'metopb'
'bh' 1.478 -> 1.748
'bl' 1.478 -> 1.748
source:
https://cimss.ssec.wisc.edu/patmosx/instrument_files/avhrr_1_instr.dat
I further checked metopa, noaa18 and noaa19 - these are correct!

ninahakansson, 2020-01-22:
Very good that you found it! Maybe that is causing all the problem with Metop-B calibration.

mraspaud, 2020-01-23:
@helgaweb do you feel like making a Pull Request for that ?

sfinkens, 2020-01-23:
I can assist of course

closed 2020-01-23 by mraspaud

helgaweb, 2020-01-23:
Might be of interest for you. The plot shows the mean of the low and high calibration slopes for the day since launch until 2019 for MetOp-A CH3A.
[patmosx_coeffs_metopb_ch3A.pdf](https://github.com/pytroll/pygac/files/4103234/patmosx_coeffs_metopb_ch3A.pdf)

PR #50 (2020-01-23): Fix typo in MetOp-B calibration
Fixes #49