pytroll/pygac #62: Computation of Earth scene radiance

helgaweb, 2020-05-20:
I was wondering, why the Earth scene radiance (Ne) does not account for linear radiance estimate Nlin as described in the KLM User's Guide. Following [7.1.2.4-6 on page 7-9](https://www1.ncdc.noaa.gov/pub/data/satellite/publications/podguides/N-15%20thru%20N-19/pdf/0.0%20NOAA%20KLM%20Users%20Guide.pdf), Ne is calculated as: Ne = Nlin + Ncor

https://github.com/pytroll/pygac/blob/2d7fbafbf03ae59d14b1fffdd33154358faee139/pygac/calibration.py#L553-L558

carloshorn, 2020-05-22:
Hi @helgaweb, indeed, to me it also looks like Nlin is missing. Currently, I am working on a refactoring of `calibration.py`, where I want to put more comments on the equations. Hope to push the corresponding commit to #58 today.

helgaweb, 2020-05-22:
@carloshorn that will be very helpful, thanks! I think the documentation in the KLM User's guide is quite well done and provides a good basis.
Alright, let's wait for @abhaydd and @mraspaud ideas on this and then I could open a PR (after you finished the refactoring).

carloshorn, 2020-05-22:
I checked the code, and it is doing the right thing, but written in a confusing way...
The coefficients are defined here:
https://github.com/pytroll/pygac/blob/2d7fbafbf03ae59d14b1fffdd33154358faee139/pygac/calibration.py#L304-L306
Which shows, that the linear correction `'b1'` term is added by one, which therefore accounts for the linear part. However, I agree that there should be a huge alert sign in the code.

abhaydd, 2020-05-22:
@helgaweb @carloshorn Yes, the Nlin is of course included properly (but implemented in an unclear way).   Otherwise we would have been off by hundreds of degrees. The non-linearity correction is typically only a fraction and translates to less than few Kelvins.

helgaweb, 2020-05-22:
@abhaydd Thanks a lot. Sorry I've overseen the change in the b1 coefficient (as clarified by @carloshorn) and was therefore confused about the calculation of Ne.

carloshorn, 2020-05-22:
@helgaweb, I pushed the comment lines to #58. I would appreciate if you could have a look at it and leave a feedback telling if the comments are sufficient to understand what the functions do.

abhaydd, 2020-05-22:
@helgaweb @carloshorn I can understand why one can get confused. The code clearly needs revamping. And it’s good to have the fresh minds like yours and Carlos’s :)

helgaweb, 2020-05-22:
@carloshorn Sure. I hope to give you some feedback by monday!

mraspaud, 2020-05-25:
I'm reopening this as I think it's a problem that it isn't documented clearly that the coeffs are taking acount of this. Hopefully #58 closes this.

PR #58 (2020-06-11): export coefficients to json file
In favour of giving more control over the calibration coefficients to the user, this PR will export the coefficients from `calibration.py` to an external data file.

Closes #62
closed 2020-06-11 by mraspaud