pytroll/pygac #74: Remove Cython dependency

carloshorn, 2020-08-06:
It would be nice to replace `correct_tsm_issue.mean_filter` with a pure python implementation.

A possible solution is:
```python
import numpy as np
import bottleneck as bn

def rolling_window(a, shape):  # rolling window for 2D array
    s = (a.shape[0] - shape[0] + 1,) + (a.shape[1] - shape[1] + 1,) + shape
    strides = a.strides + a.strides
    return np.lib.stride_tricks.as_strided(a, shape=s, strides=strides)

def std_filter(data, box_size):
    shape = (box_size, box_size)
    border = box_size // 2
    # need to surround the data with NaNs to calculate values at the boundary
    padded_data = np.pad(
        data, (border, border), 
        mode='constant', 
        constant_values=np.nan
    )
    windows = rolling_window(padded_data, shape)
    n3, n2, n1, n0 = windows.shape
    windows = windows.reshape((n3, n2, n1*n0))
    result = bn.nanstd(windows, axis=2)
    return result
```
Using bottleneck instead of the equivalent numpy function reduces the processing time to become competitive with Cython or numba.

Another possibility would be an implementation with numba:

```python
import numpy as np
import numba as nb

@nb.njit
def std_filter(data, box_size):
    N = box_size // 2
    nrows, ncols = data.shape
    std = np.empty_like(data)
    for j in range(nrows):
        for i in range(ncols):
            total = 0.0
            count = 0
            for jj in range(-N, N+1):
                for ii in range(-N, N+1):
                    j_ = j+jj
                    i_ = i+ii
                    if j_<0 or j_>=nrows:
                        continue
                    if i_<0 or i_>=ncols:
                        continue
                    value = data[j_, i_]
                    if np.isnan(value):
                        continue
                    total += value
                    count += 1
            if count == 0:
                std[j, i] = np.nan
                continue
            mean = total / count
            diff2 = 0.0
            for jj in range(-N, N+1):
                for ii in range(-N, N+1):
                    j_ = j+jj
                    i_ = i+ii
                    if j_<0 or j_>=nrows:
                        continue
                    if i_<0 or i_>=ncols:
                        continue
                    value = data[j_, i_]
                    if np.isnan(value):
                        continue
                    diff2 += (value - mean)**2
            std[j, i] = np.sqrt(diff2 / count)
    return std
```

mraspaud, 2020-08-06:
I find the numpy/bottleneck implementation cleaner.
Could you provide some numbers to get an idea of the performance?

carloshorn, 2020-08-06:
[image]

Using pure `numpy` (1300 ms) is definitely too slow, because processing an entire file with pygac-run takes about 10 s. However, `numpy + bottleneck` (272 ms) is competitive with `numba` (115 ms), which should be at a similar speed as a `Cython` implementation.

mraspaud, 2020-08-06:
That's really nice!

sfinkens, 2020-08-10:
:+1: for the numpy + bottleneck variant!

carloshorn, 2020-08-10:
A clear plus for numpy + bottleneck is that both packages have a major version greater than one. People could argue that a stable release should be a minimum requirement for production code.
Maybe in future the numba stancil feature could be an elegant solution, but for today I would also vote for numpy + bottleneck.

djhoese, 2020-08-10:
> both packages have a major version greater than one.

This is not a good reason for choosing packages in the python ecosystem, especially the scientific python ecosystem. I've argued the point that Cython needs to go to version 1.0 in the past on other forums, but they had their own timeline for things. Regardless, it looks like Cython is working on a 3.0 release: https://pypi.org/project/Cython/#history (skipping 1.0 and 2.0)

carloshorn, 2020-08-11:
@djhoese, that's true, good example for your point is the pandas library. It was the first choice for many things long before the release of 1.0 on January this year.

PR #83 (2020-09-11): Replace cython filter
This PR closes #74.
closed 2020-09-11 by mraspaud