pytroll/pygac #85: Update installation instructions

sfinkens, 2020-10-15:
The [installation instructions](https://pygac.readthedocs.io/en/develop/installation.html) are outdated and should be updated. 

- There's no need for using prefix with virtualenv or conda. Just `pip install pygac` for the latest release or `pip install git+...` for the latest development version
- Development installation instructions would be nice, too

carloshorn, 2020-11-02:
Nice, we got a hacktoberfest tag.
I could also provide some instructions on the calibration coefficient creation and the json layout.
Does pytroll have a style guide for the documentation? Otherwise, I would just try to write it as similar as possible to the existing docs.

sfinkens, 2020-11-02:
Yeah, but now it's over :scream:

Nevertheless, sounds great! I think as long as it's not too much text, we can stick to the README. But at some point we should probably change to sphinx.

PR #99 (2022-01-04): Improve documentation
Improve documentation:

- Update installation instructions
- Add usage examples
- Move algorithm descriptions to a new "Methods" chapter
- Move HDF description to a new "Legacy Output" chapter
- Add more readers and an inheritance diagram to API chapter

You can preview the updated version here: https://sfinkens.github.io/

Along the way I also included dependencies for development in `setup.py` and fixed a minor bug in `Reader.get_calibrated_channels` (using timestamps before they are computed).

Closes #85, closes #43.
closed 2022-01-04 by sfinkens