pytroll/pygac #88: Half a pixel shift on the x-axis for Metop GAC data.

mraspaud, 2020-11-11:
Disclaimer: this is probably not an issue in pygac, but because it involves pygac and because of a lack of other place, I would like to have the discussion here.

@helgaweb has brought to my attention four passes from metops A and B where there is a consistent shift in x-direction. Since these are considered KLM satellites, there is no correction applied to the navigation pygac (or satpy for that matter), so we are wondering where this can come from. @sfinkens @carloshorn any ideas? Attaching some images so you can see.

[image]
[image]
[image]
[image]

mraspaud, 2020-11-11:
For the records, these were generated using pygac output form clara a2, but I downloaded one of the original gac files from NOAA class, plotted it with satpy and pygac, and could not see any difference.

helgaweb, 2020-11-11:
Thanks @mraspaud Here are further impressions - a visual comparison of LAC L1c data by the University of Bern and PyGAC CLARA-A2 data. 
[image]
[image]
[image]

sfinkens, 2020-11-11:
Oh, that's nasty. I have no idea what the reason might be. But somebody mentioned in a meeting just now that the LAC -> GAC averaging for METOP might not be done onboard but in the ground segment. And maybe the methods differ.

helgaweb, 2020-11-11:
@abhaydd Please also have a look.

mraspaud, 2020-11-11:
@sfinkens Yes, that was one of my hypothesis, that the navigation doesn't match the aggregation in the same manner as it does on NOAA GAC.

helgaweb, 2020-11-11:
@sfinkens Yes, I looked into that. And have notes & plots on this. Even though it was new for me that the onground resampling should take place at NOAA facilities and not EUMETSAT processing facilities.

abhaydd, 2020-11-11:
Thanks @helgaweb for informing about this anomaly. I think we need to have credible/reliable information on how the processing of GAC differs between NOAA and EUMETSAT facilities (may be it doesn't). We should probably raise this point in tomorrows FCDR meeting.

helgaweb, 2020-11-11:
@abhaydd Yes indeed, sofar its an open issues and only hypothesis. I made a few tests, but didn't had proper access for test data. And as said I went into the documentation...but this is also not reliable.

mraspaud, 2020-11-12:
I was just talking to @abhaydd and he was wondering if the gac aggregation scheme wasn't causing the problem. He seems to remember that 4pixels are aggregated and the lon/lat is taken from the 5th pixel.
@helgaweb could you check where a given lon/lat in a GAC pass is located in the corresponding LAC pass, columnwise ?

abhaydd, 2020-11-12:
Do we have any statistic w.r.t. how frequently such a shift occurs? Is it a systematic feature or have we just checked few scenes? Thanks.

helgaweb, 2020-11-12:
@mraspaud It's supposed to be the same lon/lat info columnwise. But that's one of the things, why I was asking for this data earlier. So yes.

@abhaydd Few scenes were checked visually. The analysis of the relative geometric accuracy of LAC/GAC contains a much larger sample. But of course, there could be other error sources involved (even after the extensive testing).

mraspaud, 2020-11-26:
@helgaweb did you have time to look further at this?

helgaweb, 2020-11-26:
@mraspaud Yes I had. I compiled information from the KlM User's Guide and analysed few scenes (FRAC, LHRR, HRPT, in comparison to GAC from UniBe and NSS. Sorry for the slow response here. I will reach out soon.

mraspaud, 2022-10-28:
@helgaweb any news on this? we were talking about this again...

helgaweb, 2022-10-28:
@mraspaud I stopped working on that same time ago. I think I shared the latest plots with @abhaydd and you in dec 2020? I did a lot of literature search -> you find this in my CMSAF final report. I can send it to you via email? Let me know how I can support this...

mraspaud, 2022-10-28:
getting your report would be great. I can then document here.

oembury, 2025-03-19:
Hi - has there been any news with this / does the issue affect other KLM sensors?

I haven't looked at pygac geolocation in detail yet, but from `GAC_LONLAT_SAMPLE_POINTS = np.arange(4, 405, 8)` it looks like it assumes that the tie-points are in the centre of GAC pixels. Our reader assumes:

* POD: tie-points are the same as HRPT/LAC i.e. for the third FOV in the GAC pixel: i.e. `np.arange(4.1,405,8)`
* GAC: tie-points are for the "5th FOV" so are actually between GAC pixels: i.e. `np.arange(4.5,405,8)`

mraspaud, 2025-03-20:
@oembury We have not found sources describing the aggregation patterns for metop unfortunately. If you have more information, feel free to share it.

Regarding the pod and klm tie point positions, what you right here is very interesting. Would you be able to share the documentation sources for these? I'm thinking this would be great to add at least in the code and possibly here: https://github.com/pytroll/pygac/wiki/Navigation-error-in-GAC-and-LAC-data

oembury, 2025-03-20:
Sorry - we don't have any documentation on the tie-point positions, just the POD / KLM user manuals.
I think it was from a conversation with @jmittaz about 10 years ago... I'll ask if he recalls any more details, but I think we're about 30 years to late to find any documentation!

mraspaud, 2025-03-21:
Yes, that's the issue with this data, the documentation is hard to find if findable at all!
Would be great if you could get more details.

oembury, 2025-04-07:
I have tried adjusting the GAC sample points to use the "5th FOV" as used in SST CCI and it does fix the half pixel shift in a MetOp GAC file I am testing on

[image]

[image]

[image]

[image]

mraspaud, 2025-04-10:
@oembury that looks very good indeed! I think we should implement this.

mraspaud, 2025-04-11:
https://github.com/pytroll/pygac/pull/143 should correct this.