pytroll/pygac #107: Fix sun-earth distance correction

sfinkens, 2022-07-11:
Recently a bug in pyorbital's sun-earth distance correction was fixed: https://github.com/pytroll/pyorbital/issues/97. 

Pygac does not use pyorbital for this, but makes the same mistake:

https://github.com/pytroll/pygac/blob/05ba35e84caddbe96e5053b35d80abe20a091c01/pygac/utils.py#L89

I propose to use pyorbital for this.

@ninahakansson @abhaydd @kgkarl This would change the output of the FDR dataset (to the better).

mraspaud, 2022-07-11:
It makes total sense to remove code duplication, I'm in favour.

To make sure the bug doesn't reappear, maybe a test case for reflectance computation could be added at the same time?

sfinkens, 2022-07-11:
Sorry, false alarm. As pointed out by @abhaydd the factor is applied only once, so there's no problem.

https://github.com/pytroll/pygac/blob/05ba35e84caddbe96e5053b35d80abe20a091c01/pygac/calibration.py#L340

I'm opening a new issue regarding code duplication.

closed 2022-07-11 by sfinkens

mraspaud, 2022-07-11:
@sfinkens, I still think it makes sense to remove code duplication and use pyorbital here.