pytroll/pygac #140: compute_missing_lonlat errors up to 100 km

oembury, 2025-04-03:
After fixing the error in #139 I tried comparing the latitude/longitudes calculated with pyorbital and comparing them to the original geo-location in a MetOp GAC Level 1b file. There are still significant differences up to 100 km at edge of swath, while even nadir geolocation is  around 5 km out

The error is in a different direction on each side of the swath and can look like a yaw error. There is a strong dependence on latitude - could this indicate that pygac / pyorbital has an incorrect Earth shape?

[image]

[image]

[image]

[image]

[image]

Same problem affects FRAC data:
[image]

mraspaud, 2025-04-03:
That is for reporting this issue!
You may be right that it's a yaw problem, because metop uses yaw steering right?

oembury, 2025-04-03:
Yes - it is the yaw steering not being applied.

Obviously that doesn't matter for the intended use of `_compute_missing_lonlat`, but it would be useful to verify compute_missing_lonlat against the Level 1b for a MetOp sensor and check tie-point positions for #88

oembury, 2025-04-04:
Using the MetOp yaw steering corrections will fix the large shifts seen here. This does require a minor modification to pyorbital which currently only supports a constant yaw correction: pytroll/pyorbital#186 updates ScanGeometry to support per-scanline yaw in the same way as roll and pitch

mraspaud, 2025-04-10:
That PR is now merged

oembury, 2025-04-10:
You can calculate reasonable MetOp geolocation by using:
```python
rpy = np.deg2rad(reader.scans['total_applied_attitude_correction']/1e3) * [1, 1, -1]
reader._rpy = np.hsplit(rpy, 3)
lon, lat = reader._compute_missing_lonlat(reader._times_as_np_datetime64)
```
This will return latitude/longitude coordinates with errors equivalent to ~1 GAC pixel (~4km at nadir, ~20km at edge of swath) the Level 1b geolocation appears better from initial visual inspection of the coastlines.

* For MetOp `total_applied_attitude_correction` appears to be in the opposite direction to `constant_*_attitude_error`, but for NOAA-17 is was in the same direction
* pygac / pyorbital is applying the rpy corrections by rotating by roll, pitch, yaw. While the KLM User guide implies it should be -roll, -pitch, -yaw
* KLM User Guide is defining the local x,y,z differently to pyorbital. I think that means the pitch should be inverted (but not roll or yaw)

mraspaud, 2025-04-11:
Thanks a lot for investigating this @oembury !
The attitude correction has never really been used in pyorbital, and was mostly a proof of concept on my part (and I don’t know much about flight dynamics to start with), so I’m more than happy to review PRs that are fixing the issues you mention.